rust-embedded/linux-embedded-hal #5: Unable to use embedded-hal SPI traits?

chocol4te, 2018-04-03:
Hello,

I'm new, so apologies in advance for any mistakes. I'm trying to test an embedded-hal driver on a Raspberry Pi and get these errors:

``the trait `embedded_hal::blocking::spi::write::Default<u8>` is not implemented for `hal::<unnamed>::Spidev``

``the trait `embedded_hal::blocking::spi::transfer::Default<u8>` is not implemented for `hal::<unnamed>::Spidev``

My driver works on a Blue Pill board, what do I need to change for it to work with `linux-embedded-hal`?

Thanks :)

ryankurte, 2021-09-26:
hi @fmckeogh, sorry this took so long to get back to, did you get it working in the end?

usually this kind of error is because of a mismatch in hal versions, but a _lot_ has changed since 2018.

fmckeogh, 2021-09-27:
@ryankurte Haha no worries, I'm sure it must have been a version mismatch but I've long forgotten what driver this was anyway😅

Thanks though!

closed 2021-09-27 by fmckeogh