rust-embedded/linux-embedded-hal #20: Request: Replace sysfs-gpio with gpio-cdev

drinkdhmo, 2019-05-27:
Since [sysfs is now deprecated](https://github.com/rust-embedded/rust-sysfs-gpio/issues/38), we should migrate dependence to [gpio-cdev](https://github.com/posborne/rust-gpio-cdev).

posborne, 2019-05-29:
While deprecated, the reality in the field is that there are still kernels being deployed that do not yet have support for gpio-cdev.  This is increasingly changing but I wanted to provide that perspective.  I believe the last of the major changes to the cdev userspace ABI landed in 4.8.  With 4.9 LTS kernel being common for newer projects it might be safe to start pushing cdev.  There will be people running existing devices on older kernels, however, and jumping might leave those people in the cold.

That being said, for what the embedded-hal project is, I think moving to gpio-cdev (which I will revisit placing under wg maintainership) in the next half year probably makes sense and is the right move as kernels.

ryankurte, 2019-05-29:
i was wondering about this too. 
i think switching to cdev is the right move, but perhaps given the long tail of kernels for embedded hardware it might be worth including both with a feature flag or something to regress?

RandomInsano, 2019-06-02:
+1 for a feature flag. I have a few devices that are going to be locked to Linux 4.4 indefinitely.

ryankurte, 2021-09-26:
howdy, we have deployed feature flags to switch between implementations in [`v0.3.0`](https://github.com/rust-embedded/linux-embedded-hal/blob/v0.3.x/Cargo.toml) and provide both in [`v0.4.0-alpha.X`](https://docs.rs/linux-embedded-hal/0.4.0-alpha.0/linux_embedded_hal/index.html#structs) (with feature-flags to disable each as required).

so i think we can close this, thanks folks! we don't (at the moment) have a way to automagically choose at runtime, but if there is a need for this behaviour we can open a new issue / PR.

closed 2021-09-26 by ryankurte