rust-embedded/linux-embedded-hal #24: Add example?

vmasdani, 2019-10-05:
Greetings.
I was kind of having a hard time getting the grasp of trying to use the digital GPIO pin. I worked it out in the end anyway, but I think it would be cool to add basic  examples as simple as LED blinking, LED breathing, etc. like this:
```rust
extern crate linux_embedded_hal as hal;
use hal::sysfs_gpio::{Pin, Direction};
use std::error::Error;
use std::thread;
use std::time::Duration;

fn main() -> Result<(), Box<dyn Error>> {
    let pin = Pin::new(198);

    gpio.set_direction(Direction::Out)?;
    gpio.set_value(1)?;
    thread::sleep(Duration::from_secs(1));
    gpio.set_value(0)?;

    Ok(())
}
```

thejpster, 2019-10-28:
This seems like a good idea. Would you be able to add this as a PR to create a file in `./examples`, and also note which SBC this is designed to run on / has been tested on?

vmasdani, 2019-10-29:
> This seems like a good idea. Would you be able to add this as a PR to create a file in `./examples`, and also note which SBC this is designed to run on / has been tested on?

Sure, I'll make a PR later when I get my hands on the board to test it again. By the way, the SBC I tried this on was Orange Pi Zero

vmasdani, 2020-05-18:
Hi. I'm sorry I haven't responded about this issue for a long time. I've thought about making the example and I thought it would be good to make the example in `gpio_cdev` instead of `gpio_sysfs` since it is deprecated. But I tried adding the dependency in `Cargo.toml` like this:

```toml
[package]
name = "gpiotest"
version = "0.1.0"
authors = ["vmasdani <valianmasdani@gmail.com>"]
edition = "2018"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
linux-embedded-hal = { version = "0.3", features = ["gpio_cdev"] }
```

And I got this error when building with `cargo build`

```
error: failed to select a version for `linux-embedded-hal`.
    ... required by package `gpiotest v0.1.0 (/home/valianmasdani/codes/rust/gpiotest)`
versions that meet the requirements `^0.3` are: 0.3.0

the package `gpiotest` depends on `linux-embedded-hal`, with features: `gpio_cdev` but `linux-embedded-hal` does not have these features.
```

Same goes when I change the feature to `gpio_sysfs`

```
the package `gpiotest` depends on `linux-embedded-hal`, with features: `gpio_sysfs` but `linux-embedded-hal` does not have these features.
```

When I them separately though, it's running fine and I'm able to blink an LED using `set_value(value)` (I still use Orange Pi Zero):
```
[dependencies]
linux-embedded-hal = "0.3"
gpio-cdev = "0.2.0"
```
It works fine, but I ended up not being able to use `embedded_hal` GPIO functions like `is_high()` or `set_low()`.
Am I missing something here or it's actually a bug/wrong documentation?