rust-embedded/linux-embedded-hal #38: Unable to build on nightly (rustc 1.44.0-nightly (38114ff16 2020-03-21))

little-arhat, 2020-03-26:
Steps to reproduce: git clone && cargo build:

```
>>= cargo -v build
       Fresh cfg-if v0.1.10
       Fresh void v1.0.2
       Fresh cc v1.0.50
       Fresh semver-parser v0.7.0
       Fresh version_check v0.9.1
       Fresh rustc-demangle v0.1.16
       Fresh nb v0.1.2
       Fresh semver v0.9.0
       Fresh libc v0.2.68
       Fresh bitflags v1.2.1
       Fresh rustc_version v0.2.3
       Fresh embedded-hal v0.2.3
       Fresh nix v0.14.1
       Fresh backtrace-sys v0.1.35
       Fresh nix v0.11.1
       Fresh byteorder v1.3.4
       Fresh termios v0.2.2
       Fresh serial-core v0.4.0
       Fresh ioctl-rs v0.1.6
       Fresh backtrace v0.3.46
       Fresh sysfs_gpio v0.5.4
       Fresh i2cdev v0.4.4
       Fresh spidev v0.4.0
       Fresh error-chain v0.12.2
       Fresh cast v0.2.3
       Fresh serial-unix v0.4.0
       Fresh gpio-cdev v0.2.0
   Compiling linux-embedded-hal v0.3.0 (/Users/r/prj/p/linux-embedded-hal)
     Running `rustc --crate-name linux_embedded_hal src/lib.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type lib --emit=dep-info,metadata,link -C debuginfo=2 --cfg 'feature="default"' --cfg 'feature="gpio-cdev"' --cfg 'feature="gpio_cdev"' --cfg 'feature="gpio_sysfs"' --cfg 'feature="sysfs_gpio"' -C metadata=453f24c6b88d9603 -C extra-filename=-453f24c6b88d9603 --out-dir /Users/r/prj/p/linux-embedded-hal/target/debug/deps -C incremental=/Users/r/prj/p/linux-embedded-hal/target/debug/incremental -L dependency=/Users/r/prj/p/linux-embedded-hal/target/debug/deps --extern cast=/Users/r/prj/p/linux-embedded-hal/target/debug/deps/libcast-f695f5742b822139.rmeta --extern embedded_hal=/Users/r/prj/p/linux-embedded-hal/target/debug/deps/libembedded_hal-4c986ac6d091353f.rmeta --extern gpio_cdev=/Users/r/prj/p/linux-embedded-hal/target/debug/deps/libgpio_cdev-713eeea279cdabc6.rmeta --extern i2cdev=/Users/r/prj/p/linux-embedded-hal/target/debug/deps/libi2cdev-836156e2317b18cf.rmeta --extern nb=/Users/r/prj/p/linux-embedded-hal/target/debug/deps/libnb-c1350daed77af41f.rmeta --extern serial_core=/Users/r/prj/p/linux-embedded-hal/target/debug/deps/libserial_core-8f3a3c9ba21d72dd.rmeta --extern serial_unix=/Users/r/prj/p/linux-embedded-hal/target/debug/deps/libserial_unix-50fbf10e8f258cc6.rmeta --extern spidev=/Users/r/prj/p/linux-embedded-hal/target/debug/deps/libspidev-19ef1dfbba2bd214.rmeta --extern sysfs_gpio=/Users/r/prj/p/linux-embedded-hal/target/debug/deps/libsysfs_gpio-9a66359ec333fd93.rmeta -L native=/Users/r/prj/p/linux-embedded-hal/target/debug/build/backtrace-sys-af67b5ab2dfcc035/out`
error[E0432]: unresolved import `i2cdev::linux`
  --> src/lib.rs:39:13
   |
39 | use i2cdev::linux::LinuxI2CMessage;
   |             ^^^^^ could not find `linux` in `i2cdev`

error[E0433]: failed to resolve: could not find `linux` in `i2cdev`
   --> src/lib.rs:124:20
    |
124 |     inner: i2cdev::linux::LinuxI2CDevice,
    |                    ^^^^^ could not find `linux` in `i2cdev`

error[E0433]: failed to resolve: could not find `linux` in `i2cdev`
   --> src/lib.rs:133:52
    |
133 |     pub fn new<P>(path: P) -> Result<Self, i2cdev::linux::LinuxI2CError>
    |                                                    ^^^^^ could not find `linux` in `i2cdev`

error[E0433]: failed to resolve: could not find `linux` in `i2cdev`
   --> src/lib.rs:139:28
    |
139 |             inner: i2cdev::linux::LinuxI2CDevice::new(path, 0)?,
    |                            ^^^^^ could not find `linux` in `i2cdev`

error[E0433]: failed to resolve: could not find `linux` in `i2cdev`
   --> src/lib.rs:145:66
    |
145 |     fn set_address(&mut self, address: u8) -> Result<(), i2cdev::linux::LinuxI2CError> {
    |                                                                  ^^^^^ could not find `linux` in `i2cdev`

error[E0433]: failed to resolve: could not find `linux` in `i2cdev`
   --> src/lib.rs:147:34
    |
147 |             self.inner = i2cdev::linux::LinuxI2CDevice::new(&self.path, u16::from(address))?;
    |                                  ^^^^^ could not find `linux` in `i2cdev`

error[E0433]: failed to resolve: could not find `linux` in `i2cdev`
   --> src/lib.rs:155:26
    |
155 |     type Error = i2cdev::linux::LinuxI2CError;
    |                          ^^^^^ could not find `linux` in `i2cdev`

error[E0433]: failed to resolve: could not find `linux` in `i2cdev`
   --> src/lib.rs:164:26
    |
164 |     type Error = i2cdev::linux::LinuxI2CError;
    |                          ^^^^^ could not find `linux` in `i2cdev`

error[E0433]: failed to resolve: could not find `linux` in `i2cdev`
   --> src/lib.rs:173:26
    |
173 |     type Error = i2cdev::linux::LinuxI2CError;
    |                          ^^^^^ could not find `linux` in `i2cdev`

error[E0433]: failed to resolve: could not find `linux` in `i2cdev`
   --> src/lib.rs:188:27
    |
188 |     type Target = i2cdev::linux::LinuxI2CDevice;
    |                           ^^^^^ could not find `linux` in `i2cdev`

warning: unused import: `I2CDevice`
  --> src/lib.rs:38:20
   |
38 | use i2cdev::core::{I2CDevice, I2CMessage, I2CTransfer};
   |                    ^^^^^^^^^
   |
   = note: `#[warn(unused_imports)]` on by default

warning: unused import: `I2CMessage`
  --> src/lib.rs:38:31
   |
38 | use i2cdev::core::{I2CDevice, I2CMessage, I2CTransfer};
   |                               ^^^^^^^^^^

warning: unused import: `I2CTransfer`
  --> src/lib.rs:38:43
   |
38 | use i2cdev::core::{I2CDevice, I2CMessage, I2CTransfer};
   |                                           ^^^^^^^^^^^

error: aborting due to 10 previous errors

Some errors have detailed explanations: E0432, E0433.
For more information about an error, try `rustc --explain E0432`.
error: could not compile `linux-embedded-hal`.

Caused by:
  process didn't exit successfully: `rustc --crate-name linux_embedded_hal src/lib.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type lib --emit=dep-info,metadata,link -C debuginfo=2 --cfg 'feature="default"' --cfg 'feature="gpio-cdev"' --cfg 'feature="gpio_cdev"' --cfg 'feature="gpio_sysfs"' --cfg 'feature="sysfs_gpio"' -C metadata=453f24c6b88d9603 -C extra-filename=-453f24c6b88d9603 --out-dir /Users/r/prj/p/linux-embedded-hal/target/debug/deps -C incremental=/Users/r/prj/p/linux-embedded-hal/target/debug/incremental -L dependency=/Users/r/prj/p/linux-embedded-hal/target/debug/deps --extern cast=/Users/r/prj/p/linux-embedded-hal/target/debug/deps/libcast-f695f5742b822139.rmeta --extern embedded_hal=/Users/r/prj/p/linux-embedded-hal/target/debug/deps/libembedded_hal-4c986ac6d091353f.rmeta --extern gpio_cdev=/Users/r/prj/p/linux-embedded-hal/target/debug/deps/libgpio_cdev-713eeea279cdabc6.rmeta --extern i2cdev=/Users/r/prj/p/linux-embedded-hal/target/debug/deps/libi2cdev-836156e2317b18cf.rmeta --extern nb=/Users/r/prj/p/linux-embedded-hal/target/debug/deps/libnb-c1350daed77af41f.rmeta --extern serial_core=/Users/r/prj/p/linux-embedded-hal/target/debug/deps/libserial_core-8f3a3c9ba21d72dd.rmeta --extern serial_unix=/Users/r/prj/p/linux-embedded-hal/target/debug/deps/libserial_unix-50fbf10e8f258cc6.rmeta --extern spidev=/Users/r/prj/p/linux-embedded-hal/target/debug/deps/libspidev-19ef1dfbba2bd214.rmeta --extern sysfs_gpio=/Users/r/prj/p/linux-embedded-hal/target/debug/deps/libsysfs_gpio-9a66359ec333fd93.rmeta -L native=/Users/r/prj/p/linux-embedded-hal/target/debug/build/backtrace-sys-af67b5ab2dfcc035/out` (exit code: 1)

```

little-arhat, 2020-03-31:
Broken in the same way on `rustc 1.44.0-nightly (211365947 2020-03-30)`.

ryankurte, 2020-03-31:
Hmm, `git clone git@github.com:rust-embedded/linux-embedded-hal.git` then `cargo build` works okay for me on `rustc 1.44.0-nightly (f509b26a7 2020-03-18)`.

What OS are you running, is this in a workspace or anything?

closed 2020-04-07 by little-arhat