rust-embedded/linux-embedded-hal #45: Linux PREEMPT_RT

reneherrero, 2020-06-18:
Hi,
Wondering if anyone has taken at stab at using the Linux PREEMPT_RT patch with Rust to run some real-time(ish) tasks?
Thanks,

posborne, 2020-06-19:
I have experience, unrelated to my Rust work, of maintaining the PREEMPT_RT patches (back against 2.6).  By-and-large the patch has little or nothing to do with Rust; the patches change the behavior of the kernel and how it interacts with userspace but userspace applications (like those written in Rust running on Linux) do not require changes.

The only thing you might do different in a system running a kernel patched with CONFIG_PREEMPT_RT relates to how you set up scheduling priorities for your process.  That's pretty outside of the scope of anything that we would really care about here as none of this changes the fundamental APIs by which applications interact with devices from userspace.

In general, my experience has been that when doing things like SPI you can expect some decreased latencies on average with PREEMPT_RT on a loaded system.  If you have really tight constraints and hard requirements on realtime behavior you'll be better suited using a coprocessor to do the realtime work (which could be running rust) or writing a kernel driver (this doesn't provide perfect guarantees but eliminates the userspace roundtrips).  Some newer stuff in the kernel like BPF that allow for pushing of behavior into kernelspace from userspace may eventually give us some additional options for banging bits without the userspace overhead and without sacrificing safety.

reneherrero, 2020-06-19:
> The only thing you might do different in a system running a kernel patched with CONFIG_PREEMPT_RT relates to how you set up scheduling priorities for your process.

That's what I'm trying to figure out how to do.  I tried looking for something resembling the `pthread_attr_setschedpolicy` and `pthread_attr_setschedparam` methods in C but no luck.  The closest I was able to find is this: https://github.com/chocol4te/nylon/blob/master/examples/cyclic.rs

I guess I'm just surprised nobody has come out with a cleaner way to go about this

posborne, 2020-06-19:
Might be worth exploring adding wrappers for `sched_*` to nix as a start to making things more accessible.  Some of the sched functions are covered by not sched_setscheduler: https://github.com/nix-rust/nix/blob/master/src/sched.rs

reneherrero, 2020-06-19:
Yeah, the nix crate was the first place I looked.  Then I turned here thinking that more folks in this space are concerned with scheduling.

Closing this off as I think it's probably not the correct forum.

Thanks for taking the time to reply Paul

closed 2020-06-19 by reneherrero

posborne, 2020-06-19:
> Yeah, the nix crate was the first place I looked. Then I turned here thinking that more folks in this space are concerned with scheduling.

No problem, I happen to be a maintainer (though not too active recently) of nix as well.  I think looking at nix makes sense to add the safe syscall portion.

reneherrero, 2020-06-19:
Created https://github.com/nix-rust/nix/issues/1260.  Let's see what comes out of it.