rust-embedded/linux-embedded-hal #46: Update Crate Release

dmitrijoseph, 2020-07-17:
Currently trying to build with gpio-cdev feature flag like 
```linux-embedded-hal = { version ="0.3.0", features = ["gpio_cdev"] }```

Gives the error
```
error: failed to select a version for `linux-embedded-hal`.
    ... required by package `myPackage v0.1.0 (/project)`
versions that meet the requirements `=0.3.0` are: 0.3.0

the package `myPackage` depends on `linux-embedded-hal`, with features: `gpio_cdev` but `linux-embedded-hal` does not have these features.

failed to select a version for `linux-embedded-hal` which could resolve this conflict
```
Version 0.3.0 of linux-embedded-hal was released 7 months ago. gpio-cdev support was merged afterward on March 2nd.

dmitrijoseph, 2020-07-17:
Temporary Work Around
`linux-embedded-hal = { git ="https://github.com/rust-embedded/linux-embedded-hal.git", features = ["gpio_cdev"] }`

rubberduck203, 2020-08-25:
Any news on a new release?

dfrankland, 2021-01-04:
There's a new `0.4.0-alpha.0` version on crates.io now:
https://crates.io/crates/linux-embedded-hal/0.4.0-alpha.0

eldruin, 2021-03-05:
@rust-embedded/embedded-linux what is missing for a final `0.4.0` release?

ryankurte, 2021-03-05:
[`e-h@1.0.0`](https://github.com/rust-embedded/embedded-hal/issues/177) :-/ we -could- backport some of these changes to the 0.3.x series if it was useful

eldruin, 2021-03-08:
ah, I see. I would say backporting depends on users' interest in `gpio_cdev` integration, then.
If you need it, feel free to join this conversation!

2bndy5, 2024-10-14:
[v0.4.0](https://github.com/rust-embedded/linux-embedded-hal/releases/tag/v0.4.0) has been released. Should this be closed?

eldruin, 2024-10-17:
I think so. Thanks @2bndy5 !