rust-embedded/linux-embedded-hal #54: Get "-bash: ./test: No such file or directory" error on BeagleBone

psiphi75, 2020-12-01:
When I use the below configuration and cross compile to ARMv7 I get the following error message on the BeagleBone:

```sh
debian@beaglebone:~$ ./test 
-bash: ./test: No such file or directory
```

When I use v0.2.2 of linux-embedded-hal or earlier I do not get this error.  If I remove `extern crate linux_embedded_hal as hal;` from the code I do not get the error.

Any ideas?

## The code

**`src/main.rs`**
```Rust
extern crate linux_embedded_hal as hal;

fn main() {
    println!("Hello");
}
```

**`.cargo/config`**

```TOML
[target.armv7-unknown-linux-musleabihf]
linker = "arm-linux-gnueabihf-gcc"
rustflags = ["-Clink-args=-Xlinker -rpath=/usr/lib/arm-linux-gnueabihf"]
```

**`Cargo.toml`**
```TOML
[package]
name = "test"
version = "0.1.0"
edition = "2018"

[dependencies]
linux-embedded-hal = "0.4.0-alpha.0"
```

To compile I use:

```sh
cargo build --target=armv7-unknown-linux-musleabihf --release
```

caemor, 2020-12-01:
Could you please give us the output from `readelf --all ./test | grep interpreter` (from https://stackoverflow.com/questions/35071872/bash-no-such-file-or-directory). It looks like there is a problem with your linker :thinking:

psiphi75, 2020-12-02:
Yes, that's it thank you.  But only the first part of my problem.  The command `readelf --all ./test | grep interpreter` gave:

```
      [Requesting program interpreter: /usr/lib/ld.so.1]
```

So I changed my target to `armv7-unknown-linux-gnuabihf` and this resulted in:

```
      [Requesting program interpreter: /lib/ld-linux-armhf.so.3]
```

Now I'm stuck with the following message on my BeagleBone:

```
./test: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.32' not found (required by ./test)
```

I'm going remain with version 0.2.2 for now.  Thanks.

ryankurte, 2021-09-26:
in your initial `.cargo/config` it appears you're trying to build a rust `musl` target using a `gnu` linker, you need to use one or the other which should look like:

```toml
[target.armv7-unknown-linux-gnueabihf]
linker = "arm-linux-gnueabihf-gcc"

[target.armv7-unknown-linux-musleabihf]
linker = "arm-linux-musleabihf-gcc"
```

to get the musl cross tooling you can either [compile it yourself](https://musl.libc.org/), find a [pre-compiled](http://musl.cc/) package, or use [cross](https://github.com/rust-embedded/cross) which provides docker containers with the required tooling pre-installed.

> ./test: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.32' not found (required by ./test)

`gnu` targets require `libc` to be available at runtime, it looks like you don't have this installed on your beaglebone?