rust-embedded/linux-embedded-hal #64: Unable to get gpio-cdev feature working on v0.3.0

CSRaghunandan, 2021-09-20:
As the title suggests, v0.3.0 of linux-embedded-hal does not have gpio-cdev feature, but it is claimed to have in the documentaiton.

ryankurte, 2021-09-20:
huh, it looks like it's in the [`v0.3.x`](https://github.com/rust-embedded/linux-embedded-hal/blob/v0.3.x/Cargo.toml#L13) branch, but not in release [`v0.3.0`](https://github.com/rust-embedded/linux-embedded-hal/blob/v0.3.0/Cargo.toml#L10), we could probably stand to point release that. you might like to patch to `linux-embedded-hal = { git = "https://github.com/rust-embedded/linux-embedded-hal", branch = "v0.3.x" }` to see if that works as expected in the interim.

CSRaghunandan, 2021-09-20:
I see. Yeah, you guys can make a point release with the cdev feature enabled.

Until then, I can use the code from 0.3.x branch.

commit 72af1092 (2021-09-27): Merge #66

66: Updates for v0.3.x r=ryankurte a=eldruin

Fixes #64
PR #66 (2021-09-27): Updates for v0.3.x
Fixes #64