rust-embedded/linux-embedded-hal #72: CAN trait implementation

marcelbuesing, 2021-12-09:
I think it would be great to add implementations for the new embedded-hal CAN traits via [socketcan](https://www.kernel.org/doc/html/latest/networking/can.html). Would you be open for a PR for this?

eldruin, 2021-12-09:
That would definitely be interesting!

marcelbuesing, 2021-12-13:
Ok I basically created a fork of the [socketcan](https://crates.io/crates/socketcan) crate, you can find the fork [here](https://github.com/marcelbuesing/socketcan-rs/tree/embedded-hal-1.0.0-alpha.6). Sadly the original crate is no longer maintained but works quite well for me and still sees significant download numbers on crates.io. I reduced the dependencies and removed modules which were not necessary for this use case. A simple example is [here](https://github.com/marcelbuesing/socketcan-rs/blob/embedded-hal-1.0.0-alpha.6/examples/simple.rs). It already works but still needs some minor cleanup.

My suggestion would be to move this project/fork copy to the rust-embedded organization and basically move all the code that is currently in the `embedded_hal_impl` module to the linux-embedded-hal crate referencing the socketcan crate in the rust-embedded organization, analogous to `gpio-cdev` or `sysfs_gpio`. If this an option for the rust-embedded org would this be ok for your @mbr ? I see great potential in e.g. getting started with the embedded-hal socketcan implementation and then e.g. migrating to bxcan in projects.

ryankurte, 2021-12-14:
hey thanks for your effort! if @mbr (cc. @fpagliughi) are up for handing `socketcan` over it's a :+1: from me, any other opinions @rust-embedded/tools?

adamgreig, 2021-12-14:
(maybe @rust-embedded/embedded-linux ?)

nastevens, 2021-12-29:
Seems logical to me as well :+1:

marcelbuesing, 2022-02-10:
Ok since there is no response from the socketcan maintainers, how about a new crate name: `linux-socketcan`.

fpagliughi, 2022-02-10:
Hey All. Apologies. I agreed to share maintainer responsibilities for the socketcan crate last year, then got pretty sick missed a few months of work, and let my opensource projects go dormant for a while. But I've been catching up recently, and have a CAN project starting soon, so can begin contributing.