rust-embedded/linux-embedded-hal #80: Enable i2c as a feature?

zuckschwerdt, 2022-04-01:
The dependency on `i2cdev` for `i2c` is currently the only thing tying this to linux. I confirmed this crate to compile and work on unix with `i2c` removed.
Would it be an option to offer `i2c` as a (default enabled) feature?

ryankurte, 2022-04-01:
seems reasonable to me! though it may be a breaking change for folks selecting `gpio` implementations at the moment, maybe both `spi` and `i2c` so we're consistent?

zuckschwerdt, 2022-04-02:
True, adding an `i2c` feature would break `default-features = false` (for `gpio` selection) setups.
Alternative could be conditional compilation on `target_os`, it would just-work and not expose settings. But there might be users wondering where `i2c` is (on host != target).

eldruin, 2022-04-02:
I would be fine with adding `spi` and `i2c` features. We can do so in the upcoming 0.4.0 release, since it is breaking anyway and this does not seem time-critical to me.
Have you tested SPI to work on non-linux platforms? also, which platform are you working with?

zuckschwerdt, 2022-04-03:
Not time-critical at all, just ergonomics to me. I use modular hardware, missing components are expected. Building with inaccessible or missing hardware interfaces is fine for me. In my application it turns out to be less involved to just recognize a missing /dev/spidev, GPIO or serial at runtime then to stub out all the code.

zuckschwerdt, 2022-04-05:
For anyone wanting to test this, there are branches for 0.3 and 0.4 to show the changes:
- https://github.com/zuckschwerdt/linux-embedded-hal/tree/feat-i2cspi03
- https://github.com/zuckschwerdt/linux-embedded-hal/tree/feat-i2cspi04

eldruin, 2022-04-05:
Nice, would you create a PR with your feat-i2cspi04 branch?

commit 12847c4b (2022-04-06): Merge #81

81: Add features for spi and i2c r=eldruin a=zuckschwerdt

Adds the features `i2c` and `spi` for an optional dependency on `i2cdev` and `spidev`.

Discussion in #80.
Closes #80
PR #81 (2022-04-06): Add features for spi and i2c
Adds the features `i2c` and `spi` for an optional dependency on `i2cdev` and `spidev`.

Discussion in #80.
Closes #80
closed 2022-04-06 by bors[bot]