rust-embedded/linux-embedded-hal #82: I2C transaction breaks I2c trait contract

KaSroka, 2022-05-18:
According to I2c trait transaction contract is as follows:
```
Transaction contract:
- Before executing the first operation an ST is sent automatically. This is followed by SAD+R/W as appropriate.
- Data from adjacent operations of the same type are sent after each other without an SP or SR.
- Between adjacent operations of a different type an SR and SAD+R/W is sent.
- After executing the last operation an SP is sent automatically.
- If the last operation is a `Read` the master does not send an acknowledge for the last byte.
- `ST` = start condition
- `SAD+R/W` = slave address followed by bit 1 to indicate reading or 0 to indicate writing
- `SR` = repeated start condition
- `SP` = stop condition
```

Using following simple test code:
```rust
use embedded_hal::i2c::blocking::{I2c, Operation as I2cOperation};
use linux_embedded_hal::I2cdev;

fn main() {
    let mut i2c_dev = I2cdev::new("/dev/i2c-1").unwrap();

    let data = [0xaa; 1];
    let mut ops = [I2cOperation::Write(&[0x40]), I2cOperation::Write(&data)];
    i2c_dev.transaction(0x3d, &mut ops).unwrap();
}
```

I'm expecting no SR "repeated start condition" to appear as both operations are of the same type (Write). Unfortunately I can see SR "repeated start condition" on the bus as well as SAD+R/W "slave address followed by bit 1 to indicate reading or 0 to indicate writing" (positions 6 and 8). This was captured using logic analyzer on embedded Linux board for the sample code attached above:
[image]

eldruin, 2022-08-04:
Interesting find!
Following the code path you can see that [two linux write messages are created](https://github.com/rust-embedded/linux-embedded-hal/blob/66e773bda9af92069287c1d27cc0b8a6e81b3ec1/src/i2c.rs#L127) and [sent](https://github.com/rust-embedded/linux-embedded-hal/blob/66e773bda9af92069287c1d27cc0b8a6e81b3ec1/src/i2c.rs#L134) to the underlying [rust-i2cdev](https://github.com/rust-embedded/rust-i2cdev) library.
There it is [received and the message addresses filled](https://github.com/rust-embedded/rust-i2cdev/blob/efb109011b0a37a3d5329ce9ead478d7fd1f6bf0/src/linux.rs#L289) and then [sent](https://github.com/rust-embedded/rust-i2cdev/blob/efb109011b0a37a3d5329ce9ead478d7fd1f6bf0/src/linux.rs#L293) to the [kernel via FFI](https://github.com/rust-embedded/rust-i2cdev/blob/efb109011b0a37a3d5329ce9ead478d7fd1f6bf0/src/ffi.rs#L448).
The [kernel documentation](https://www.kernel.org/doc/html/latest/i2c/dev-interface.html) states:
> The function will write or read data to or from that buffers depending on whether the I2C_M_RD flag is set in a particular message or not. **The slave address** and whether to use ten bit address mode **has to be set in each message**, overriding the values set with the above ioctl’s.

So the code path seems to be correct and this is just how the linux kernel works: it sends the messages without a stop in between, although it does send a restart with the same address for each message regardless of the type of the last one.

In order to uphold the `transaction` contract what we could do is to consolidate operations of the same type that follow each other into a single message for the kernel here.
This needs to be done for `transaction_iter` as well, but the `_iter` part of it is internally already collected and not immediately sent so it should be doable.

ohunter, 2024-05-03:
The `I2C_M_NOSTART` flag does fix the issue, but it assumes that the i2c device supports it. This can be checked using the `I2C_FUNCS` operation of `ioctl`. One potential issue with the fix is that it needs to be implemented in the `rust-i2cdev` repo as the `i2c_mgs.flags` field is private. If we can assume that the `LinuxI2CDevice::transfer` call should always fulfill the contract then this isn't an issue at all.

eldruin, 2024-05-03:
I am not opposed to make `rust-i2cdev` fulfill the contract of `embedded-hal::i2c::I2c`. However, it is possible to set the `NO_START` message flag entirely from this crate by using `with_flags()`. i.e. `LinuxI2CMessage::write(w).with_flags(I2CMessageFlags::NO_START)`.
One would need to play a bit with the chain [here](https://github.com/rust-embedded/linux-embedded-hal/blob/66e773bda9af92069287c1d27cc0b8a6e81b3ec1/src/i2c.rs#L123), though, in order to set it only when appropriate.