rust-embedded/linux-embedded-hal #86: serial-unix crate is unmaintained

samgoldman-bionautlabs, 2022-08-22:
We have noticed that this crate uses serial-unix which has not been maintained in 5 years: https://github.com/dcuddeback/serial-rs. 

Instead the maintained crate serialport (https://github.com/serialport/serialport-rs) could be used.

qrasmont, 2023-03-14:
Hi,

This means `Serial::open` should become a wrapper around `serialport::TTYPort::open` instead of `serial_unix::TTYPort::open`.

Is it then ok to modify the `Serial::open` signature to match the arguments expected by `serialport::TTYPort::open` ? Or should the existing signature be preserved ?

eldruin, 2023-03-15:
Keeping the signature is not a must. It would be nice to keep if possible, though, just because it is currently very straight-forward by simply providing a path.
We have a breaking release in the pipeline so if you are interested, now it would be a great time to have a go at it.

qrasmont, 2023-03-16:
The main difference is that `serial-rs` would rely on termios to get the baud rate while  `serialport-rs` requires the user to provide it to be able to open the port.

So the signature could be maintained by using termios.

commit 1f97fd4b (2023-03-20): Merge #89

89: serial: Replace the serial-rs crate with serialport-rs r=eldruin a=qrasmont

Replace the unmaintained serial-rs crate with serialport-rs.
Closes #86
PR #89 (2023-03-20): serial: Replace the serial-rs crate with serialport-rs
Replace the unmaintained serial-rs crate with serialport-rs.
Closes #86
closed 2023-03-20 by bors[bot]