rust-embedded/linux-embedded-hal #112: impl embedded_hal::digital::Error for linux_embedded_hal::gpio_cdev::Error

SpieringsAE, 2024-02-09:
I'm trying to write an initialisation function for a driver, but the fact that the pin initialisation functions don't give me an embedded-hal Error trait is being a bit of a pain in the behind for me.
I can't properly incorperate the errors that arise when setting up gpios in my drivers error enum.

My situation is an SPI device that has two additional gpios, a reset and an interrupt pin.
This is a very hardware specific driver so the pins and spidev are created in the driver::new() function.

The linux_embedded_hal::spi::SPIError does implement embedded_hal::spi::Error so that one is fine I think.

Is this a workable idea?

reitermarkus, 2024-02-10:
Would the implementation in https://github.com/rust-embedded/linux-embedded-hal/pull/110 solve your issue as well?

In there, the `CdevPin::new` method returns a `CdevPinError` defined in this crate instead of needing to construct the `CdevPin` from the external type as is currently the case.

SpieringsAE, 2024-02-10:
i think yes that could very much do that