rust-embedded/linux-embedded-hal #120: SPI `TransferInPlace` operation is unsound.

reitermarkus, 2025-06-12:
The implementation for the `TransferInPlace` operation violates the aliasing rule by creating an immutable reference from a mutable one, but keeping both.

https://github.com/rust-embedded/linux-embedded-hal/blob/1ffb268a71b2b5157461685cc62b325a86e41f8c/src/spi.rs#L216-L222

There needs to be a corresponding method for `SpidevTransfer`, e.g. `read_write_in_place`.

jamesmunns, 2025-10-21:
I definitely concur with this being UB! Would be great to have this fixed, we've notified @sirhcel, who will be taking a peak when possible :)

jannic, 2025-10-21:
This could be fixed with #123, but needs a release of `spidev` first, to implement the required `read_write_in_place` function.
(with https://github.com/rust-embedded/rust-spidev/pull/49)